ttinoco/PFNET #1: Printout created by the C-Functions is not shown in Jupyter

martinzellner, 2016-04-26:
The printf calls inside the C-part of the PFNET Library are going directly to stout and can therefore not be displayed in the Jupyter Notebook.

This is related to this issue:
http://stackoverflow.com/questions/35745541/how-to-get-printed-output-from-ctypes-c-functions-into-jupyter-ipython-notebook

ttinoco, 2016-04-26:
Hi Martin,
Interesting. I should be able to write C routines that return strings and
make the python wrapper get then and print them. I will do this and let u
know so we can try on jupyter.

ttinoco, 2016-04-29:
I have changed the code to have python do the printing (for the most important "show" functions, e.g., net show_components, net show_properties, contingency show and problem show) (branch c-output). I will try to get setup with jupyter and test it

martinzellner, 2016-04-29:
I had to include a `decode('UTF-8')` for python 3 compatibility since the C-output is a bytestring (Commit:  807e46c)

Seems to work fine now:
[image]

martinzellner, 2016-04-29:
Graphs are now also displayed inline with the new `jpview()` function.

[image]

ttinoco, 2016-04-29:
cython did not like some the encode statements. I had to do some work arounds; see Commit 47ef350 . 
both python2 and python3 compile, nosetests works but nosetests3 has some errors related to relative imports so I could not test python3

ttinoco, 2016-04-29:
Branch python-3-encode-fix works now for python2 and python3. All unittest pass. I had to change a bit the package structure to get this to work and make the unittests python2 and 3 compatible. Now nosetests and nosetest3 require a "python setup.py build_ext --inplace" before execution. @martinzellner you can integrate this into your python-3 branch (if you still need to do something) or we can integrate directly into master

martinzellner, 2016-04-29:
I'm fine with integrating it directly into the master.

ttinoco, 2016-04-29:
Ok, I may combine view and jpview before doing that

martinzellner, 2016-04-29:
I already did that. I will push it tonight.

ttinoco, 2016-04-29:
Oh sweet!!

closed 2016-05-04 by ttinoco