ttinoco/PFNET #5: Graph should display Bus IDs

martinzellner, 2016-05-09:
It would be nice if the Graph renders the Bus IDs as node labels.

ttinoco, 2016-05-09:
Ok. I left this out since this did not look nice on huge graphs. I can make it an option. I believe now one can only set node attributes (see tables 8,9 in http://www.graphviz.org/doc/libguide/libguide.pdf) for all nodes but not for individual ones. The pfnet routine is graph.set_nodes_properties. I will create graph.set_node_property so one can set the "label" attribute of a specific node.

ttinoco, 2016-05-09:
I have added graph.set_node_property. This routine allows setting graphviz-specific node attributes. Must be used before "set_layout". See updated visualization sample code (ex8.py)
[https://github.com/ttinoco/PFNET/blob/master/python/examples/ex8.py](url)
[image]

closed 2016-05-09 by ttinoco

martinzellner, 2016-05-09:
Looks really nice! Thanks!