ttinoco/PFNET #6: FIX constraints are not working as expected for Loads

martinzellner, 2016-05-15:
I am experiencing problems when setting the load variables as fixed. The number of fixed variables increases, however the `A` matrix seems unchanged.

Looking at https://github.com/ttinoco/PFNET/blob/master/src/problem/constr/constr_FIX.c it seems like the FIX constraint is not implemented for loads. Is this the case or am I doing something wrong?

ttinoco, 2016-05-15:
I will check this today in a few hours.

ttinoco, 2016-05-16:
I just added this and pushed to master. I added unittests but do some sanity checks of your own.

martinzellner, 2016-05-16:
Cool, thanks!

ttinoco, 2016-05-16:
No problem. Just close this issue after you see it working well in your code.

martinzellner, 2016-05-16:
The load constraints work now as expected. Thanks!

closed 2016-05-16 by martinzellner