ttinoco/PFNET #11: NET_COST should not penalise the battery power

martinzellner, 2016-05-16:
If NET_COST also penalises the battery power (e.g. the injection from the storage), it cannot be beneficial to use the storage.

ttinoco, 2016-05-16:
the nest cost function penalizes battery charging (because this increases net consumption) and rewards battery discharging

ttinoco, 2016-05-16:
Should we be considering variable prices? and battery charging and discharging efficiencies?

martinzellner, 2016-05-16:
I am considering variable prices. As soon as I penalise the battery power, the battery is not used anymore. I have to think about that a bit more...

ttinoco, 2016-05-16:
The question is whether in your current problem there is a benefit or shifting consumption

martinzellner, 2016-05-16:
With variable prices, it should be beneficial to charge at low prices and to discharge at high prices, right?

ttinoco, 2016-05-16:
Yes. You should do some experiments with your problem to make this happen.

martinzellner, 2016-05-16:
Without the penalty on the battery power:
[image]
With the penalty:
[image]

Without the penalty the result seems to make a lot more sense to me. On the other hand, I agree that the penalty should be correct. 

I will play around with it a bit more and see if i get to a satisfactory result..

martinzellner, 2016-05-16:
I think I figured it out. The NET_COST function also rewards the slack bus as a generator leading to an unbounded problem. This can be evaded by setting the energy price of the slack bus to zero...

closed 2016-05-16 by martinzellner