ttinoco/PFNET #16: pythonic interface

ttinoco, 2016-06-11:


awig, 2016-12-17:
[Adding my email response to @ttinoco ]

Sorry, I did see it and was thinking of a response but got sidetracked and forgot.

I like the use of strings, definitely an improvement, but I was trying to think of a way that it could be done with the string constants instead of numeric ones.  My other thought was to maybe pass a generic attribute of a network element class, like `pfnet.Bus.vmag` for instance to refer to voltage magnitude.   I'm not sure that makes sense though or would even be considered pythonic.

It would be good to list out all the options so that we could discuss the names. For instance, instead of `voltage magnitude` what about `vmag` or having both?

I'm continuing to think about it some more.  We should chat next week sometime, I want to catchup and start discussing additions from the EPRI work that you do and do not want merged back.

ttinoco, 2016-12-17:
Thanks for the comments.  A key question I have, which is related to your comment, is that whether full names or abbreviations should be used . Abbreviations should probably match the attribute names of a class. For example, one could have 'v_mag' so that it matches Bus.v_mag. Right now I am leaving more towards full names.  The current "string maps" I am using are here:
https://github.com/ttinoco/PFNET/blob/pythonic-pfnet/python/pfnet/cstrings.pyx

awig, 2016-12-17:
That sounds good to me. All the mappings currently listed look good as well.

Let me know when it is ready for testing.

ttinoco, 2016-12-19:
@awig The changes are ready for testing. The pythonic-pfnet branch uses strings in the python wrapper for component types, flag types, component quantities, function type, constraint types. No more strange constants (e.g. BUS_PROP_NOT_SLACK) and logical ORs at the python level. The unittests and the sphinx tutorial have been updated. Docs are not live but need to be created using "make html".  The dependent package GRIDOPT has also been updated to work with this version of pfnet in the pythonic-pfnet branch. Let me know if these changes in pfnet and gridopt are ok for you and then I can merge them to dev.

awig, 2016-12-20:
@ttinoco 
Should I keep OPTALG and raw-parser on 'dev' while testing?

ttinoco, 2016-12-20:
yes

awig, 2016-12-20:
Thx

**Testing on Mac OSX**
[x] - C PFNET builds and all tests passed
[x] - Python PFNET builds and all 65 nosetests passed
[x] - Python documentation builds with ``make html DYLD_LIBRARY_PATH=$PFNET/lib/``
   - see proposed changes in commit 0ea70dc
[x] - testing python PFNET examples in documentation
   - I recommend that there be a summary table of all elements and their quantities and properties and some explanation about what qualifies as a property, quantity. In the API documentation the properties and quantities don't really stand out and blend in with the headings. Something like in the flag types would be a bit more "readable". Note I did not read through *everything*, but mainly just tested the examples (although with a different case).
   - An example of using the ``store_sensitivities`` could be nice. 
[x] - GRIDOPT builds and all 3 nosetests passed
[x] - interactively tested that all solvers (`DCPF`, `DCOPF`, `NRPF`, `AugLPF`, and `AugLOPF`) are able to solve (I found that the IEEE300 case has a bad problem that I will need to look into).

This is all I could think to test. If there is anything else though let me know.

ttinoco, 2016-12-20:
Many thanks. I agree with your suggestion about improving the documentation of quantities, properties. I will open another issue (one closes, another opens). Sensitivities and mismatch constants still need to be turned into strings.  I tested IEEE300 but did not have issues.

closed 2016-12-20 by ttinoco

awig, 2016-12-20:
Sorry, I should clarify. I am trying with the IEEE300 'raw' case.