ttinoco/PFNET #24: Branch flows and clarification on naming

awig, 2016-08-22:
I will be adding branch flows to the branch struct and will perform the calculation of the branch flows within the struct as well. These flows will then be used by the update_properties in net.c. This will NOT change constr_PF.c.  

I will also change the naming conventions to clarify what elements of the branch and from what bus the flows are being calculated.

ttinoco, 2016-08-22:
Sounds good.  It would be useful if you let me know your plans a bit ahead of time so we can start a discussion. I am guessing you will start first with changing the naming convention.

awig, 2016-08-22:
Yes. Naming convention first and then integrating it although I think I would do just one PR to dev here.  Here is what I'm proposing for the naming:

 constr_PF.c

-         flowP —>  Pvkvm
-         flowQ —>  Qvkvm
-         flowP_sh —>  Pvkvk
-         flowQ_sh —>  Qvkvk

 elsewhere (net.c/branch.c)

-         flowP —>  P_series_bus_from, P_series_bus_to
-         flowQ —>  Q_series_bus_from, P_series_bus_to
-         flowP_sh —>  P_shunt_bus_from, P_shunt_bus_to
-         flowQ_sh —>  Q_shunt_bus_from, Q_shunt_bus_to
-         *new* P_bus_from, P_bus_to
-         *new* Q_bus_from, Q_bus_to

@ttinoco what do you think?

ttinoco, 2016-08-22:
I have feedback for these. Give me a day.

awig, 2016-08-22:
Sure. No problem.

ttinoco, 2016-08-23:
@martinzellner gave me a great suggestion today: we should change the branch field names (e.g. bus_from to bus_k) and add the new interface routines (get_bus_k, get_Pkm, etc) but KEEP the old interface routines (e.g. get_bus_from) and mark them as DEPRECATED (so that old code does not just break)

awig, 2016-08-23:
Agreed. I will include these changes as part of this issue.

awig, 2016-08-23:
For consistency should we use an underscore to separate the subscript in all cases (at least going forward)?  I notice that other fields like 'P_max' and 'v_mag' for example always separate the subscript from the identifying symbol. Thus, I would change to P_km, b_k, b_m etc.

awig, 2016-08-23:
Style question: should the old interface routines be modified to just use the new fields or call the new routines?

ttinoco, 2016-08-23:
Your subscript suggestion is good.  What do you mean by b_k and b_m? What are these variables? Do you mean bus_k and bus_m? With regards to style, either way would be ok but I am leaning towards "call the new routines"

awig, 2016-08-23:
I think calling the new routine looks cleanest, so I'll go with that.  

In branch.c their are fields for the shunt conductance and susceptance as viewed from both k and m sides (currently b_from, b_to, g_from, g_to which I would change to b_k, b_m, g_k, g_m respectively, as well as the names of the branch list objects from_next, to_next to k_next, m_next respectively.

ttinoco, 2016-08-23:
Ok, I remember now. Thanks. Should we use "next_k"? instead of "k_next"?

awig, 2016-08-23:
Yes! that reads better.

awig, 2016-08-24:
Quick update: 
I believe I have implemented all the name changes and have put place holders for the BRANCH_get_Pkm etc.  Everything compiles and I can reference the new and old naming structure.  I may have some questions about how the vector of var_values is used but need to spend more time understanding it.

awig, 2016-08-30:
Testing on Romcon/PFNET and will do a PR here after some testing.

ttinoco, 2017-02-11:
Issue has been taken care of in latest pull request

closed 2017-02-11 by ttinoco