ttinoco/PFNET #25: test_func_LOAD_UTIL nosetest fails

awig, 2016-08-24:
@ttinoco 

The following nosetest fails on my mac and on Nick's: 

======================================================================
FAIL: test_func_LOAD_UTIL (tests.test_functions.TestFunctions)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/ajwigington/projects/Power/PFNET/python/tests/test_functions.py", line 1252, in test_func_LOAD_UTIL
    self.assertLessEqual(error,EPS)
AssertionError: 2.914487117911174 not less than or equal to 2.0

----------------------------------------------------------------------
Ran 63 tests in 19.018s

I think the is a matter of just making this assertion a little less stringent. Not sure what is the source of the difference though.

ttinoco, 2016-08-24:
I agree. It could be due to numerical precision and the generation of different random numbers on different computers (despite the same seed? is that possible?)

ttinoco, 2016-08-24:
The seeding is done at the beginning of a test. However, a test loops through the cases present in the data directory and this can be different for different users (hence the sequence of random vectors used for finite-difference checks of derivatives may vary). For example, I have some .raw files that do not come with pfnet when you checkout. I will relax the tolerance of the tests a bit and improve the safeguards for divisions by small values.

ttinoco, 2016-11-25:
This has been fixed in the multi-period branch

closed 2016-11-25 by ttinoco