ttinoco/PFNET #29: Import fails on multi_period branch

martinzellner, 2016-08-29:
The import of PFNET currently fails on the [multi_period branch](/ttinoco/PFNET/tree/multi_period):
```
ImportError: dlopen(/usr/local/lib/python3.5/site-packages/pfnet/cpfnet.cpython-35m-darwin.so, 2): Symbol not found: _BAT_get_E_final
  Referenced from: /usr/local/lib/python3.5/site-packages/pfnet/cpfnet.cpython-35m-darwin.so
  Expected in: /Users/murphy/git/PFNET/lib/libpfnet.so
 in /usr/local/lib/python3.5/site-packages/pfnet/cpfnet.cpython-35m-darwin.so
```

ttinoco, 2016-08-29:
For me it works. Maybe you are linking to the old library?

martinzellner, 2016-08-29:
Weird. It seems to matter whether you are in module directory of PFNET when you execute the import.
Never mind, it works now.

ttinoco, 2016-08-29:
Yes, it matters. When importing pfnet in the pfnet/python directory python the local package has precedence over the system pacakage and the local package does not have the cython stuff built in-place. You should do "python setup.py build_ext --inplace"

martinzellner, 2016-08-29:
Makes sense. Obviously I got lucky a couple of times.
Maybe we should change that in the [documentation](http://ttinoco.github.io/PFNET/python/start.html#installation).

ttinoco, 2016-08-29:
It says in the docs to use build_ext already. Do you have something else in mind? It would be better I think to have the "install" command also build inplace.

martinzellner, 2016-08-29:
I think it would be good to also use build inplace here:
[image]

ttinoco, 2016-08-29:
Ok I see. However, maybe is better to incorporate this build inplace in the install command, to save users some trouble. Having to do two commands is also a bit annoying during development

closed 2016-10-03 by ttinoco