ttinoco/PFNET #32: Testing: Multi-period

awig, 2016-11-17:
## Multi-period Testing

Testing with:

- PFNET `multi_period`
- raw-parser `multi_period`
- OPTALG `dev`
- GRIDOPT `multi_period`

### Installation

uninstalling all above packages first from virtualenv

#### Mac OSX

##### PFNET and RAW_PARSER

1. delete all files in PFNET/lib folder
2. cd to raw-parser
3. set `$PFNET`
4. [x] `make clean && make` for raw-parser completes and places `librawparser.so` into PFNET/lib
5. cd to PFNET
6. [x] `make clean && make NO_GRAPHVIZ=1 NO_RAW_PARSER=1` 
7. [x] `make clean && make` 
8. cd to PFNET/python 
9. [x] `python setup.py clean && python setup.py install --no_raw_parser --no_graphviz`
10. [x] `python setup.py clean && python setup.py install`
11. note: still have to do `python setup.py build_ext --inplace` and add the PFNET/lib to `$DYLD_LIBRARY_PATH` to get nosetests to run
12. [ ] `nosetests -v` fails

```
(env)ajwigington@adams-air:~/projects/Power/PFNET/python$ nosetests -v
test_constr_BOUND (tests.test_constraints.TestConstraints) ... FAIL
test_constr_DCPF (tests.test_constraints.TestConstraints) ... FAIL
test_constr_DC_FLOW_LIM (tests.test_constraints.TestConstraints) ... FAIL
test_constr_FIX (tests.test_constraints.TestConstraints) ... FAIL
test_constr_GEN_RAMP (tests.test_constraints.TestConstraints) ... ok
test_constr_LBOUND (tests.test_constraints.TestConstraints) ... FAIL
test_constr_LINPF (tests.test_constraints.TestConstraints) ... FAIL
python(17315,0x7fff73f04000) malloc: *** error for object 0xd000000000000000: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6
```

  - [x] import works fine.
  - [x] Example `pfnet.Network(5)` works fine
  - [x] Example should show loading network `net.load('ieee14.mat')`. 
    - I recommend including a `load` in the documentation example
  - [x] All the examples in the documentation work as expected.
  - what PF file is the example using when it shows output?
  - if you could add external (e.g., vargen). what happens if you define different number time-periods for different elements?
  - in the documentation (not sure if it belongs here or in the solver) but that the problem is solved for all time periods at once?

I will add more as I continue testing.

ttinoco, 2016-11-19:
* what PF file is the example using when it shows output?
good catch, should be ieee14

* if you could add external (e.g., vargen). what happens if you define different number time-periods for different elements?
specifying time periods should not be allowed. It is currently not an argument in add_vargens

* in the documentation (not sure if it belongs here or in the solver) but that the problem is solved for all time periods at once?
this should be in gridopt docs

With regards to your nosetest fails:
I am confused about how you are building the python wrapper. You build pfnet.so without graphviz and raw_parser then you are not passing --no_garphiz nor --no_raw_parser when you are doing build_ext. 
You also have 2 consecutive "python setup.py install" with and without these flags. 

I tested multi_period before and everything worked fine. All unittests passed. However, I tried it right now and I am getting a whole bunch of errors of the form
` RuntimeWarning: tp_compare didn't return -1 or -2 for exception` 
I had never seen these errors and the code has not changed since I last ran the unittests. After some google searches I see that these errors might be python bugs. I will let you know as soon as I figure them out.

ttinoco, 2016-11-19:
@awig Update:
Something changed in numpy and that was the cause of the new errors 
`TypeError: object of type 'AttributeInt' has no len()`
This has been fixed in the last commit to multi_period.
All unittests pass for me.  Let me know if things work on your systems.

ttinoco, 2016-11-25:
Any updates on this?

awig, 2016-11-28:
@ttinoco Sorry it was Thanksgiving Holiday and I was still working on the EPRI report.

## Mac OSX 10.11.

I updated PFNET to commit: 829bc1e 
I am using numpy  version 1.11.2, which version are you using?

The building of the libpfnet.so works fine still, installing the python library also works.
```
export PFNET=~/PFNET
cd $PFNET
make clean && make
...
cd python
python setup.py install
```
Then it still works as expected when I run anything.

FYI: I was only testing the graphviz and raw-parser options separately so ensure they work earlier.

I still get the nose tests errors when I run:
```
python setup.py build_ext --inplace
nosetests -v
```

My version of nose is 1.3.7.

## Ubuntu 16.04
Everything works properly including the nosetests.  It must be an issue with Mac OSX. I'm looking into it but haven't had luck yet.

ttinoco, 2016-11-28:
Thanks for the update. My numpy version is  1.11.2 and my nose version is 1.3.1

awig, 2016-11-28:
I was testing gridopt with the multi-period support and keep getting a bad problem error.  I'm simply trying to run it with the 5 load.P changes. 

```
import pfnet as pf
import numpy as np
import gridopt

net = pf.Network(5)
net.load('data/ieee14.mat')

for load in net.loads:
   load.P = np.random.rand(5)
net.update_properties()

method = gridopt.power_flow.new_method('NRPF')
method.solve(net)
```
I tried an AugLPF solver too but both fail to solve.

My suggestion is to add in the documentation a multi-period case that you are sure will work.  Maybe something like:
```
scale = np.array(map(lambda x: x/100.0, range(100, 110, 2)))
for load in loads:
    load.P = load.P[0] * scale
```

It would be nice to have a simple way to import the load and generation changes over a time period with a given format if we could find one say from a csv or excel file. I've only ever seen this type of data in simple csv file formats given the name of the substation.  We would have to settle on a format first. What do you think?

ttinoco, 2016-11-28:
The solvers fail to solve or you get a "bad problem" exception? 

What about
```
for load in net.loads:
   load.P = load.P + 0.2*np.random.rand(5)
```

You actually do not neet update_network. That is only to update the properties that are functions of the data (e.g., maximum voltage, mismatches, etc)

About the csv output. I think it is a nice idea. We can do this in general about many properties of the network.

ttinoco, 2016-11-28:
I am creating an issue for adding a multi-period example in the docs

awig, 2016-11-28:
It fails to solve.

Your suggestion for the 0.2 scaling works for me on both NRPF and AugLPF. 

Agreed. It should definitely be on gridopt. Do you want me to start an issue there?

ttinoco, 2016-11-28:
I started it.

awig, 2016-11-28:
Okay. I see that now.

I'll try to figure out the nose tests issues on MacOSX before I push back to ``dev`` branches.

ttinoco, 2016-11-28:
Sounds good. Many thanks for the help.

awig, 2016-11-28:
My pleasure!

awig, 2016-11-30:
An update. I spent a good amount of time on this with no luck. Neither the ``dev`` nor the ``multi_period`` branch work on Mac OSX. The version on ``ROMCON/PFNET/master`` still does. The tools install fine and seem to work with the basics I've tested.

I've looked into running the nosetests verbosely but don't see any significant differences before it fails. 

I tried cythonizing cpfnet.c before installing and building the cpfnet.so extension with an updated build and even trying to update to setuptools instead of distutils since we are using setuptools on romcon. None of those changes worked.

Any suggestions on how I should try debugging the tests further?  I'm going to next look into how to run nosetests with a debugger.

Any suggestions for debugging the nosetests?

ttinoco, 2016-11-30:
I have never done it. Is nosetests still giving you a bunch of "FAIL" and "malloc error"?
Are your versions of cython/python/numpy/scipy the same between ubuntu and mac osx?
I just built and tested the multi_period combination (raw_parser,pfnet,gridopt) in CentOS release 6.8 and it worked.

awig, 2016-11-30:
Yes. That one still.

Good point I'll check numpy and scipy versions.

ttinoco, 2016-11-30:
One thing I do is try to focus on one test at a time: e.g., nosetests  -s -v tests.test_network
Then I add print statements before the an assert that fails to inspect what is going on. I am sure there are much better ways though.

awig, 2016-11-30:
I made sure they were all the same versions for libraries that I could think would be important.

environ | Python | Cython | Numpy  | Scipy  | Nose
:------ | ------ | ------ | ------ | ------ | -----
OSX     | 2.7.12 | 0.25.1 | 1.11.0 | 0.17.0 | 1.3.7
Ubuntu  | 2.7.12 | 0.25   | 1.11.0 | 0.17.0 | 1.3.7

I'll start running through the single tests.

awig, 2016-11-30:
Aw!!! Finally realized what happened.  I had Windows versions of the data files on my mac in the *PFNET/data* folder, which was causing the problem. Since we fixed this in ROMCON's PFNET parsing then it worked there.  

Okay so it runs now and all tests pass.
```
(env2)ajwigington@Adams-Air:~/projects/Power/PFNET/python$ nosetests
.................................................................
----------------------------------------------------------------------
Ran 65 tests in 44.718s

OK
```

Is there anything you want me to merge now before you merge ``multi_period`` back into  ``dev`` and I close this issue?

awig, 2016-11-30:
@ttinoco Do you think it worthwhile to add a section on testing the C library?  Demonstrating using *run_tests.out*?

ttinoco, 2016-11-30:
Is there anything you want me to merge now before you merge multi_period back into dev and I close this issue?
 **No, just close this issue. I will then merge multi_period into dev.** 

Do you think it worthwhile to add a section on testing the C library? Demonstrating using run_tests.out?
**This would be nice. In the Readme?**

awig, 2016-11-30:
Yes, in the Readme. Let me know if you want me to do it.

Closing this issue.

closed 2016-11-30 by awig