ttinoco/PFNET #36: Support Qmax=Qmin=0 of a generator

awig, 2017-01-20:
I found this problem while merging the romcon additions with the multiperiod and pythonic improvements from ttinoco. Note this data should be considered bad but exists for some test cases like the ieee14 raw file and could potentially occur in real cases as well.

adam:

> I found the error occurs because the slack generator has Qvar max of 0 and Qvar min of 0. In the .mat the Qvar max is 10.0. I tried just changing this to a very small value it it still runs fine. I have  implemented a temporary solution to set the default Qvar max to a small, epsilon like value in the meantime.

tomas:

> Some constraints in pfnet add some epsilons to prevent dividing by zero when Qmin = Qmax.

> If it is related to Q we can start looking into the voltage regulation constraints by generators and in the PV-PQ switching heuristics (all in pfnet). Actually, we can start looking into the generator reactive power participation constraints. I think the error should come from there. in line 118 of nr.pf, where the maximum is throwing an error, you can check whether the error comes from norminf(fdata.f) or norminf(fdata.r). If it comes from fdata.r, then it comes from the linear constraints and hence it likely comes from Q participation constraints (fdata.f is related to nonlinear constraints, e.g. voltage regulaation)

awig, 2017-01-24:
I noticed that an older version of PFNET for generators with Qmin=Qmax=0 are not considered regulating, but in more recent versions they are. In the older versions this ends up causing errors with setting variables and things for the problem. 

The old PFNET commit that I'm using:

     8f67e48eb4de546826565b8e25dfae5ccdfb100b ../PFNET (v1.0-320-g8f67e48)

I am working on asking some more experienced system studies EPRI people how they think it should be handled, as I can't find a good explanation in PSSE documentation.  WOuld you please explain to me your reasoning on how it is handled?

ttinoco, 2017-01-24:
I remember that in PSSE raw files, a generator regulates a voltage
magnitude if Qmax > Qmin

https://bitbucket.org/tomastinoco/raw-parser/src/ca0d6497258157fefac805aff14033f786b84a0a/src/RAW_parser.c?at=dev&fileviewer=file-view-default#RAW_parser.c-3572

It seems that this is also the case in .mat files

https://github.com/ttinoco/PFNET/blob/dev/src/parser/parser_MAT.c#L378

This is in the parser. The parser then sets up the "reg_bus" of a
generator, which otherwise is NULL. In pfnet a generator is considered
regulating if reg_bus is not NULL.

tomas

awig, 2017-01-24:
I just remembered that I added that small amount when Qmax=0 so that it is always the case. So I'll need to remove that and resolve this issue correctly after some more thorough thought and clarification on how to handle Qmax=Qmin=0.

I also see that in RAW_net.c (below) that the regulated bus number will return the connected bus number if it is set to 0, which does agree with PSSE's manual. 

```
int RAW_GEN_get_rc_bus_number(RAW_Gen* gen) {
  if (gen->rc_bus_number == 0)
    return gen->bus_number;
  else
    return gen->rc_bus_number;
}
```

However, this could potentially be misleading when it is a non-regulating bus.

ttinoco, 2017-01-24:
After some more thinking, It is strange that Qmax=Qmin=0 is causing a
problem because they should not be regulating generators and hence should
not get involved in the "generator participation" constraints nor in the
"voltage regulation" constraints.
Tomas

awig, 2017-01-24:
Agreed. There must be some use of it that is trying to refer to all generators thinking that all of them should have a regulator bus when that is in fact incorrect implementation and needs a check first. I'll have to ask Bob more about his use.

awig, 2017-01-24:
After talking with a couple of people here, it seems they are likely using this to model a wind or solar plant that only has real power generation and provides not regulation.  
Therefore, it should definitely not have ``reg_bus``.  I'll just focus on the original problem stated on the first comment.

ttinoco, 2017-04-26:
I see you have added a check for Qmax = 0 in the raw parser (latest PR) and are adding a small number. I do not think this is a long term solution so I will not incorporate it yet until we figure what the real issues are. I still do not understand.

ttinoco, 2017-05-18:
We need to change unittests to check Qmin <= Qmax for regulating gens and make sure all relevant constraints can deal with Qmin=Qmax.

awig, 2017-05-24:
The issue I have with v32 files, wasn't an issue at all. It is this issue and I just did not check my v32 file close enough to see that the Qmin=Qmax in it while my v32 file changes Qmin to an arbitrarily small value.

I looked at the ``pfnet.Problem`` created with the Qmin=Qmax case and noticed that one of the ``x`` primal variables is ``nan`` in both a NRPF and AugLPF problem.  Just to be sure I created the project to check if it was generator reactive power ``net.get_var_projects('generator', 'reactive power')`` and when I multiplied it by ``x`` sure enough I got the ``nan`` value.

The problem stems from when the Q value for the generator is first calculated. As I checked the problematic generator and Q was set to `nan`.  This is strange because the initial Q value is set to something. I'll continue to look into whether it is parser or network setup related.

awig, 2017-05-25:
I think I found the issue in net.c.NET_adjust_generators on line 294-295. Here it calculating values to update Q for buses regulated by generators. 
[net.c line 293](https://github.com/ttinoco/PFNET/blob/master/src/net/net.c#L293)

This is my best guess right now.

awig, 2017-05-26:
@ttinoco 

I have made the following changes to **raw-parser** 

* if the bus is a slack bus and if |Qmin-Qmax|<=1e-3 MVA, then set Qmin=-9999, Qmax=+9999 (defaults and corresponds to how commercial software handles it). In this case the values entered for Qmin, Qmax must just be junk anyways because there is no way your slack bus have such unreasonable limits.

 This solved the issue, I also tested when Qmin=Qmax=0 and those generators are correctly set as non-regulating generators.  

I have implemented it with a threshold bounds of 1e-3 p.u. that will determine whether or not a generator should be regulating. If you prefer I can remove this.

I will open a new issue  there and create branch so that you can merge into dev as you have time. I will still try to add checks on all Qmax-Qmin (when a denominator), so I'll leave this issue open.

ttinoco, 2017-05-27:
I am not too sure your change in raw_parser is the best thing to do:
1) the issue can come up in other parsers so we would like to make pfnet more robust and not break
2) it is strange that when |Qmin-Qmax| is "way too small" then we make it "way too big". We are making a change that takes things to the other extreme. I would only do this if this is the explicit convention of the v32 files. 

I checked pfnet a bit
1) gen Q participation has a safeguard for Qmin=Qmax
2) generator voltage regulation constraint does not consider slack buses and hence does not consider slack generators
3) I saw net adjust_generators routine and there can be a problem there when Qmax=Qmin. However, this method was needed before by the old version of AugL that required an initial feasible point with respect to the linear constraints. I do not think it is needed anymore since the new AugL algorithm can start with infeasible points (it does not work in the nullspace of the linear equality constraints).

awig, 2017-05-30:
> I am not too sure your change in raw_parser is the best thing to do:

This particular issue will only come up because this particular erroneous data have been entered for Qmin,Qmax (= 0). By default Qmin=-9999 and Qmax=9999 for the commercial program,  but would have to check if these are the values that are actually set.  It in fact does the following check according to the documentation

> The Newton-Raphson based
> power flow solutions require that the difference between the controlling equipment's high and low
> reactive power limits be greater than 0.002 pu for all setpoint mode voltage controlling equipment...

Regardless,I this would just be modeling it as somebody else does it doesn't mean it is the best way.

These are my thoughts on how to handle it:  

1.  Raw files only, Qmin=Qmax=0 for a slack generator.   I do think this can be considered bad data and the default values used for raw files (although we could change these values.) This can occur during parsing in raw-parser.

2. Any type of file, |Qmin-Qmax|<=some small pu value for a slack generator. Provide the user a warning and try to remove this generator as a slack, error if it is the only slack. (we could also check the available slack values in the system as a whole as a "pre-screening"?)

3. Any type of file, |Qmin-Qmax|<= some small pu value for a PV generator. Provide the user a warning and convert the generator to a PQ supplying generator using the given Q value. 

4. If `NET_adjust_generators` is still needed/wanted (I tried a search and couldn't find any actual uses) we can just handle the issue the same way you have in the constraints.

@ttinoco what do you think?

ttinoco, 2017-05-30:
I do not agree with "remove this as a slack" or "convert the gen to a PQ". PFNET should work by using safegurads like max{ Qmax-Qmin, epsilon} so why would we need to make more complicated changes? I am also not sure about issuing warnings since using a small epsilon would not compromise results. We should try removing NET_adjust_generators from PFNET and from NRPF and AugLPF in gridopt, and check that the power flow still perform as usual. We can always add this function back later if the need comes up.

awig, 2017-05-30:
Okay. I think it is a question of whether to trust the data. 

I'll try removing `NET_adjust_generators`.

ttinoco, 2017-05-31:
I think we should give warnings when there is bad data such as Qmin > Qmax, but I think Qmin = Qmax should be fine and handled with safegurads.

ttinoco, 2017-12-19:
Should we close this?

awig, 2017-12-19:
Right now, if I set Q_max,Q_min = 0 of a generator in the raw file both raw parsers (C and Python) create networks with generators that have Q_max,Q_min still at 0.

PFNET still has NET_adjust_generators and I don't believe it is still in gridopt. 

I think we can close it for now, and reopen it if needed (if we find this to be an issue in the large real cases).

ttinoco, 2017-12-20:
So I set Qmin=Qmax and net.adjust_generators created a bunch of "nans". I added a safeguard and added unittests that checks that no nans are present and that the generator participation constraints are satisfied after net..adjust_generators, which is the purpose of net.adjust_generators. This has been integrated in the dev branch.

closed 2017-12-20 by ttinoco

awig, 2017-12-20:
Ok. Thanks