ttinoco/PFNET #51: setuptools and pypi

ttinoco, 2017-03-13:
"pip install pfnet" should install C library and python wrapper. 
Seems like solution might be customizing setuptools command using cmdclass and calling configure,make,etc with subprocess call.

awig, 2017-03-15:
- do you want to submit both Python 2.7 and 3.5/6 versions?
- there will be a few things that need to be added it seems to fulfill the pypi requirements
for reference:  https://wiki.python.org/moin/CheeseShopTutorial

ttinoco, 2017-03-15:
@awig 
Yes, that would be great. I have been reading a bit and it does not seem easy to create a pfnet pypi package that would automatically install the C library. It is possible but not pretty due to the fact that our python package is a subfolder of the C library package. The other option is to just install the wrapper, but that does not make too much sense.

ttinoco, 2017-08-06:
@awig 
Got pfnet to work on pypi test site. If you can please have a look at the pypi branch (in particular manifest.in, setup.py, and build_lib.sh in python dir) and let me now what you think.

The command to install pfnet from the pypi test site is:
`sudo -E pip install -i https://testpypi.python.org/pypi pfnet`
The -E option preserves environment variables in order to detect raw_parser (see configure.ac update and pypi_pfnet branch on bitbucket raw_parser)

It should work for linux and osx. With your cmake and an extra "if statement" it should be easy to make it work on windows.

ttinoco, 2017-08-13:
This has been improved to install C library locally and use $ORIGIN in linux and corresponding trick in osx, and has been integrated in dev branch. Need to update docs and upload to pypi instead of testpypi

closed 2017-08-13 by ttinoco