ttinoco/PFNET #56: active distribution grid mutliperiod ACOPF

ttinoco, 2017-04-05:


awig, 2017-04-06:
This is very helpful.  We could also link these to modeling capabilities to be added, for instance FACTS or other power flow control devices, HVDC, voltage-dependent or complex loads, so we know what is needed to model what new network components. What do you think?

Shouldn't limiting control actions be included in constraints?

Here are a couple of other thoughts to consider

to variables:
- voltages as polar coordinates ?
- what would be needed for power flow controls such as FACTS ?

to objectives:
- minimize network losses
- minimize transfers (over specific corridors)
- maximize reactive power capability and hence voltage stability security ? 

to constraints:
- voltage dependency of loads
- voltage stability limits (not sure on the formulation, need to ask Bob)
- polar coordinate voltage limits ?

ttinoco, 2017-04-06:
Sounds good.  For limiting control actions we still have the sparsity-inducing penalty. 
I think you mean rectangular coordinates. Great idea. I would suggest moving this list of items to another more general issue called "new variables, functions and constraints". The items you have listed here are great. I will then add to that list things like: generator commitment variables, minimum up-down constraints, generator start-up costs, etc (for unit commitment).

ttinoco, 2017-05-04:
done

closed 2017-05-04 by ttinoco