ttinoco/PFNET #66: Use build_ext existing commands instead of using argparse for incdir, libdir, libpfnet

ttinoco, 2017-04-26:
python setup.py currently defines new arguments --libdirs --incdirs and --libpfnet using argparse. This is silly because build_ext already has these comments.

ttinoco, 2017-04-26:
@awig I will take care of this while processing the PR. I will provide a setup.cfg with default include dirs, libdirs and libraries (-lpfnet), and anyone can just change these or overwrite them by passing aguments to build_ext (they already exist, we don't need argparse)

ttinoco, 2017-04-27:
@awig This has been done. Please have a look at 
https://github.com/ttinoco/PFNET/blob/master/python/setup.py (pretty clean right?)
https://github.com/ttinoco/PFNET/blob/master/python/setup.cfg
and
https://github.com/ttinoco/PFNET/blob/master/python/docs/build.sh
This last script shows you an example of how to do a custom build by passing arguments to build_ext and overwriting the values specified in setup.cfg. I am using this for readthedocs but you can use this as an example for your custom build. You can also pass --libraries=xxxx if your pfnet library has another name.

closed 2017-04-27 by ttinoco

awig, 2017-05-13:
@ttinoco nice. very clean. I'm not familiar with setup.cfg as part of distutils but after some quick searching I agree it is the better approach and is also supported in setuptools.

I need to test the latest build and merge it back into romcon soon. I'll let you know if I run into an problems when building cross platform (Windows), but that is unlikely.