ttinoco/PFNET #88: for branch_get_P_km and related methods, should t come before values for consistency?

ttinoco, 2017-06-02:


ttinoco, 2017-06-02:
@awig 
what do you think?

awig, 2017-06-02:
@ttinoco 
Certainly in favor of it if it improves consistency. I was comparing it to other network element functions like [BUS_set_price(Bus* bus, REAL price, int t)](https://github.com/ttinoco/PFNET/blob/master/include/pfnet/bus.h#L239) where the value being set comes before the `t` argument.
However, i do see the consistency it would have with other functions that use arguments of type `Vec` like [FUNC_LOAD_UTIL_eval_step(Func* f, Branch* br, int t, Vec* var_values)](https://github.com/ttinoco/PFNET/blob/fb83671e43f759745eae5aed888a5f425feb4c33/include/pfnet/func_LOAD_UTIL.h#L24). Is this a good example of what you mean?

ttinoco, 2017-06-02:
Yes, exactly I was thinking of function and constraint eval step routines.
I am not sure which option  is has the majority