ttinoco/PFNET #92: Segmentation fault when slack gen contingency is applied to AugL

awig, 2017-06-15:
Using the ieee14.mat file when i try to solve using ``gridopt.power_flow.AugL()`` when applying outage to the lone slack generator at bus 1. 

```
import pfnet, gridopt
net14 = pfnet.ParserMAT().parse('./data/ieee14.mat')
augl = gridopt.power_flow.AugL()
cg1 = pfnet.Contingency()
cg1.add_gen_outage(net14.get_bus_by_number(1).generators[0])
cg1.apply()
augl.solve(net14)
>>> Solver: augL
>>> ------------
>>>  k      phi      pres      dres      gLmax     dmax    alpha    sigma   theta    code   vmax  vmin   gvdevgQvio    tvvio    svvio  
>>>  0   2.12e-02  2.32e+00  9.97e-02  4.08e+01  0.0e+00  0.0e+00  4.3e-04 1.0e+00   ----   1.09  0.0e+00  0.0e+00  0.0e+00  0.0e+00 
>>> Segmentation fault: 11
```

I have also tried with NRPF and it results in a ValueError due to nonsquare A matrix passed to the linear solver.  (I'm using scipy's superlu). 

@ttinoco How can we address this? 1) include a required check before applying a contingency that there still be at least 1 slack generator. 2) provide a check and throw a pfnet error on `pfnet.Problem.analyze` or `pfnet.Problem.eval` that checks the COO elements don't access (if that is what is happening although I don't know how to verify if that is in fact the issue).

ttinoco, 2017-06-15:
For NRPF the "nonsquare error" makes sense. We easily catch this and provide a nicer message either in gridopt after the problem is created or in optalg NR when a problem is received (we can compare the A.shape[0]+f.size to the size of x.

For AugL, this segfault error is unacceptable and we have to find what is accessing memory incorrectly. COO elements of a matrix is possible but I would not start there. I would start looking at the contingency code and make sure that it does the right thing for slack buses (is everything clear correctly?). Maybe there is a pointer inside a slack generator or inside a slack bus that is not cleared correctly. 

If the contingency code seems correct, then we can test each constraint and function used to form the power flow problem individually and see which one leads to segfault. 

Do you have a test script that I can use to debug?

After fixing this we should include this script as a unittest.

ttinoco, 2017-10-06:
I just checked and this error no longer occurs in the latest dev branch

closed 2017-10-06 by ttinoco