ttinoco/PFNET #99: revisit formulation of branch thermal limits

ttinoco, 2017-08-07:
Do we really need 2 per branch?
Try 2 per branch only when shunt impedance at both sides are different

awig, 2017-08-07:
Doubtful. I just included it for completeness. I think the values can also be different with phase shifters but I'm not certain.

ttinoco, 2017-09-25:
This issue appears to have been resolved after changing the branch flow limit constraints from:

current_magnitude - slack = 0
0 <= slack <= thermal_rating

to 

current_magnitude - slack = 0
-thermal_rating <= slack <= thermal_rating.

It is possible that in the original formulation, the lower bound barrier of the slack was putting too much restrictions in the initial steps of the algorithms. I will check with @MBaltzinger to verify ipopt and inlp performance improvements

closed 2017-09-25 by ttinoco