ttinoco/PFNET #114: add missing unittest for all recently-exposed "set" methods

ttinoco, 2017-09-03:
in branches, gens, shunts, etc (all) 
@awig I believe you are working on these (I just merged romcon-dev info ttinoco-dev).

awig, 2017-09-04:
Yes. I need to add these.

awig, 2017-09-12:
I see you most of the set functions for setting component attributes like `shunt.b_values` and `gen.Pmin` have been added.

What I think needs to be tested now are the network array allocations and adding "connections" between components (e.g., `gen.reg_bus` and `bus.add_generator()`). @ttinoco  is this correct?

ttinoco, 2017-09-12:
@awig array allocation and "connections" should be tested in a new test that uses these and other routines to re-create a network in python and then compare it to the original. You are right by saying many of the "setters" have already been tested. They are typically in test_network.py under test_buses, test_branches, test_generators, etc. It would be good to check which ones are missing.

ttinoco, 2017-09-22:
@awig Has this been done?

awig, 2017-09-22:
@ttinoco 
No. I will start it after I finish the logging on GAT.

awig, 2017-09-29:
created branch `network-allocation-tests` for this issue:

- [ ] allocation tests for branches, buses, loads, generators, shunts
- [ ] Buses
   - [ ] setters (if needed)
   - [ ] connections?
- [ ] Loads
   - [ ] setters (if needed)
   - [ ] bus connection
- [ ] Generators
   - [ ] setters (if needed)
   - [ ] bus connection
   - [ ] regbus connection
- [ ] Shunts
   - [ ] setters (if needed)
   - [ ] bus connection
   - [ ] regbus connection
- [ ] Branches
   - [ ] setters (if needed)
   - [ ] k,m bus connections
   - [ ] regbus connection

ttinoco, 2017-09-29:
@awig Can all these be achieved with a **single** test that creates a network in Python from another network, and then invokes compare_networks?

ttinoco, 2017-10-11:
Can I close this?

awig, 2017-10-12:
Yes. thanks. covered in the new test for creating a new network from one
created via parsing

ttinoco, 2017-10-16:
awesome, thanks

closed 2017-10-16 by ttinoco