ttinoco/PFNET #117: move tests/utils.py to pfnet/tests/utils.py

awig, 2017-09-06:
@ttinoco I tried to find a standard way of including some tests but did not find any that worked and of course we really don't want any of the test_xxxx.py files in the pfnet package.

https://stackoverflow.com/questions/14454744/does-it-make-sense-to-install-my-python-unit-tests-in-site-packages

Thus, i'm moving utils.py into pfnet/tests/utils.py  but we could also consider changing the name to something to help distinguish between tests/ and pfnet/tests (could be confusing for those unfamiliar with python packaging).  Maybe something like pfnet/checks or pfnet/help_tests/ or pfnet/share ?

ttinoco, 2017-09-06:
@awigs, thanks for the link. I think we can leave it as pfnet/tests/utils.py. Another one I am ok with is pfnet/test_utils/utils.py

ttinoco, 2017-09-19:
this has been done in dev branch

closed 2017-09-19 by ttinoco