ttinoco/PFNET #131: pypi on mac osx

awig, 2017-10-02:
pypi fails to build pfnet on Mac OSX because the mach-o libpfnet.dylib that is build "name" is the full path to the temporary directory where it is initially built because configure is built with --prefix=$PWD/build I believe and the $PWD is the temporary directory. Thus when the linker is called (libtool on mac ox) cpfnet.so gets linked to the full temporary path name of libpfnet.dylib which is deleted after setup finished.  

There are a couple of options, 
1) create/modify a new shell script that is mac osx specific and used add linker options -headerpad_max_install_names (recommended by apple) and -install_name to force it to use rpath (not sure if this will work but it should)
2) run mac's install_name_tool and change the path of the libpfnet shared library directly if on mac osx (might be in the build shell script or in setup.py (?))

@ttinoco  do you have a preference?

ttinoco, 2017-10-02:
Thanks. I would prefer option 3) remove prefix in ./configure and make the
build script do "make install" so that library goes to somewhere standard
like /usr/local/lib (we used to have it like this, but manually). However,
before exploring this I would like to try a few things. Right now
"setup.py" builds cpfnet.so by linking with
"./lib/pfnet/build/lib/libpfnet.so", but then it relies on
./pfnet/libpfnet.so during execution. The "loader path" is used for the
latter. I think we can try using the latter for the build as well so we can
try changing or removing "library_dirs" in setup.py. Let me know what you
think. I will do a bit more research.

(I like option 3 more than 1 and 2 because I think that with 1,2 we can
have problems related to using pfnet from setupt.py build-ext inplace vs
install)

awig, 2017-10-02:
thanks. I'll try this out tonight

ttinoco, 2017-10-03:
@awig, You said that option 2 worked. For this option, did you use @loader_path in one of the arguments of "install_name_tool"? Can you show me the command? I think that if you are using @loader_path or something similar with "install_name_tool", then this option should be good to go. I would put this command inside the "build_lib.sh" script. Also, could you make sure that it works both for install as well as for build_ext --inplace?

awig, 2017-10-03:
@ttinoco 

So, I've played around some more.  Better than running install_name_tool would be to pass the following to the ./configure command.

``
  ./configure --prefix=$PWD/build LDFLAGS="-Wl,-install_name,@loader_path/libpfnet.dylib"
``

The install_name_tool command can also work but is much wordier and longer, but I haven't been able to figure out how to change it from build_lib.sh before cpfnet.so is built. I tried the following, which changes the libpfnet.dylib install names but doesn't change how it gets linked into cpfnet.so. These commands look like 
``install_name_tool -id "@loader_path/libpfnet.0.dylib" build/lib/pfnet/libpfnet.0.dylib``

But, in order to change how libpfnet is linked in cpfnet.so I have to still run this command after
``install_name_tool -change "currentpath/libpfnet.0.dylib" "@rpath/libpfnet.0.dylib" cpfnet.so``

If we want something to work with the setup.py, the only thing I can think of would be to add an argument to the build_pfnet.sh that we can than pass from call(setup.py <args>) this could be CFLAGS or LDFLAGS.

ttinoco, 2017-10-03:
> from call(setup.py ) this could be CFLAGS or LDFLAGS.

@awig, thanks.
I like the option of using
./configure --prefix=$PWD/build LDFLAGS="-Wl,-install_name,@
loader_path/libpfnet.dylib"
in the "build_lib.sh" script. I will do this in a few hours and test it on
travis ci.
I will give you an update after.

ttinoco, 2017-10-03:
It worked! I updated the build script and tested with gridopt travis build on osx. @awig many thanks!!!

closed 2017-10-03 by ttinoco

awig, 2017-10-03:
Excellent! Glad to help