ttinoco/PFNET #144: increase network flexibility

ttinoco, 2017-12-08:
- [ ] allow adding new objects to the network. For example, users should be able to create a new Generator(), set its properties, connect it to a bus. Then, new components can be integrated into the network using Network routines, e.g., network.add_generators. These routines will take a list of components, reallocate arrays and integrate the new components
- [ ] allow removing objects from the networks. Network will have routines that take lists of components and removes them from the network.
- [ ] allow networks to create subnetworks, for example by specifying a list of buses. An option of this would be to preserve generators and loads, i.e., move them to the nearest bus on the subnetwork

@awig Can you think of other useful functionality?

awig, 2017-12-19:
Trying to catchup on things... a few ideas to through out:

- would the 3rd item include a function for creating subnetworks based on electrically disconnected portions of the overall network?
- adding areas to the pfnet network. the majority of the functionality could be handled by higher level packages, but it must to have areas (and maybe even owners) at a lower level for creating networks and equivalencing neighboring networks. For instance, some commercial solvers allow you to specify different tolerances for different areas so that one can specify lower precision in external areas.
- network equivalencing and/or disconnecting by areas. Depending if we add areas, could we implement network equivalencing algorithm(s) that would allow you to equivalence external areas?

Also, what is does OBC in particular?

This is all I can think of for now.

ttinoco, 2017-12-20:
I think the items you mention are useful. The answer to your first question is yes. I think we should open issues for "areas, owners and a way to specify "weights"" for controlling accuracy. Network equivalencing can be part of this issue but maybe is better to treat it separately as it is likely very involved. Thanks!

awig, 2017-12-20:
Sounds good to me. Yeah. I agree network equivalence-ing should be a separate issue. I can open a separate issue for it.

ttinoco, 2018-01-02:
features available in branch "flexible_net"

closed 2018-01-02 by ttinoco

awig, 2018-01-03:
Nice! I'll have to play around with it soon.