ttinoco/PFNET #150: improve regulation consistency

ttinoco, 2018-01-02:
generators can regulate bus v mag
branches can regulate P flow, Q flow and bus v mag
shunts can regulate bus v mag and Q flow

right now gens, branches, shunts do not have consistent methods for checking this. Gens have "is_regulator", branches have "is_tap_changer_v", etc, and shunts have "is_switched_v". Gens "is_regualtor" and shunt "is_switched_v" both return true of reg_bus != NULL, but branch "is_tap_changer_v" checks the branch type and hence setting/deleting reg_bus has no effect, making it a potential source of issues

awig, 2018-01-03:
Looks good to me.  I would only propose considering xxx.is_X_regulating() instead of xxx.is_X_regulator().  

Another thought for consideration would be if loads can be voltage (and maybe Q) regulating? I've heard of some large loads adding control devices on their end to reduce costs and on the distribution side DER voltage regulation, but maybe then you would just model it with equivalent components. Perhaps we can talk to Eknath and Anish about it next week. 
E.g.: http://www.njcleanenergy.com/files/file/Renewable_Programs/ACE%20Voltage%20Regulation%20-%20for%20NJBPU%2009212012.pdf

ttinoco, 2018-01-03:
> ACE%20Voltage%20Regulation%20-%20for%20NJBPU%2009212012.pdf

Cool. Thanks. I like the idea of adding regulation capabilities to other
components such as loads and var generators.