ttinoco/PFNET #175: need generator function that encourages Q_par

ttinoco, 2018-03-30:


awig, 2018-03-30:
Add participation constraint so that AugL and other solvers can encourage Q participation similar to NR. 

Constraint form:
(alpha_i * Q_j - alpha_j * Q_i)^rho = 0 
where alpha is participation
rho is integer
but generalized to m participating generators

ttinoco, 2018-03-30:
Comments
* the costraint already exists, we need a penalty function
* rho should be 2

<https://github.com/ttinoco/PFNET/issues/175#issuecomment-377639980>, or mute
the thread

awig, 2018-03-31:
Doh. Of course. I've updated the above comment

ttinoco, 2018-03-31:
ok, we basically need a function of the form
(1/2) sum (over reg buses k) sum (over pairs (i,j) of gens regulating  bus k) (alpha_i Q_j - alpha_j Q_i)^2
If there are n gens regulating bus k, there should be n-1 terms in the second summation. For example if regulating gens are gens 1,2,3, then we need (1,2), (2,3)