ttinoco/PFNET #180: Build fails on GCC 7 due to buffer overflow checks

awig, 2018-05-14:
Testing build with Ubuntu 18.04 which has  gcc version 7.3.0 (Ubuntu 7.3.0-16ubuntu3)  by default.

GCC7 has added restrictions on compiler flags for buffer overflows on sprintf. 

Snippet of error:
```
(env2) adam@romcon1:~/romcon/PFNET$ make
Making all in src
make[1]: Entering directory '/home/adam/romcon/PFNET/src'
/bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../include/pfnet    -I../include/pfnet/.. -I/home/adam/romcon/raw-parser//include -g -O2 -Wall -Werror -MT libpfnet_la-bat.lo -MD -MP -MF .deps/libpfnet_la-bat.Tpo -c -o libpfnet_la-bat.lo `test -f 'net/bat.c' || echo './'`net/bat.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../include/pfnet -I../include/pfnet/.. -I/home/adam/romcon/raw-parser//include -g -O2 -Wall -Werror -MT libpfnet_la-bat.lo -MD -MP -MF .deps/libpfnet_la-bat.Tpo -c net/bat.c  -fPIC -DPIC -o .libs/libpfnet_la-bat.o
In file included from net/bat.c:14:0:
net/bat.c: In function ‘BAT_get_json_string’:
../include/pfnet/../pfnet/json_macros.h:50:16: error: ‘%s’ directive writing up to 99 bytes into a region of size 92 [-Werror=format-overflow=]
   sprintf(temp,"\"%s\":\"%s\"",field_name,field);   \
                ^
net/bat.c:537:3: note: in expansion of macro ‘JSON_str’
   JSON_str(temp,output,"name",bat->name,FALSE);
   ^~~~~~~~
In file included from /usr/include/stdio.h:862:0,
                 from ../include/pfnet/../pfnet/bat.h:14,
                 from net/bat.c:11:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:33:10: note: ‘__builtin___sprintf_chk’ output between 10 and 109 bytes into a destination of size 100
   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       __bos (__s), __fmt, __va_arg_pack ());
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:712: recipe for target 'libpfnet_la-bat.lo' failed
make[1]: *** [libpfnet_la-bat.lo] Error 1
make[1]: Leaving directory '/home/adam/romcon/PFNET/src'
Makefile:399: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1
```

Note running `make -CFLAGS="-Werror=format-overflow=0"` results in make completing the build.

awig, 2018-05-14:
This error occurs because in JSON_str function the temp value for writing the output uses the same BUFFER_SIZE as the original char array (e.g., name[BAT_BUFFER_SIZE]) but it is adding additional chars for the json string (e.g., "name:<name[BAT_BUFFER_SIZE]"). 

- [ ] add a extra buffer for the json string element

@ttinoco should I just add the size for the extra characters for the temp string's size in the implementation or do you want a more elegant solution, like a JSON_BUFFER_SIZE_EXTRA constant?

(there are other gcc7 errors that occur once I've resolve these)

ttinoco, 2018-05-15:
Thanks @awig, I will take a look as soon as possible and will follow up
here.

ttinoco, 2018-06-11:
> PR created. I'll go ahead and create a separate one for the strict aliasing

Great! Thanks