ttinoco/PFNET #197: Allow user to set ratings for thermal constraints

awig, 2018-11-02:
Right now AC, DC and linear AC all use ratingA for their limits.  Desire to allow setting to B and C as well. 

Would this be better done as (1) setting an "active rating" parameter associated with the Branch element that can be changed with a method along the lines as "set_active_rating"? or (2) set when constructing the contraint of the particular type?  I prefer idea (1) at the moment.

ttinoco, 2018-11-02:
The way to do this is to take advantage of constraint parameters. We can
add a parameter to the constraint that imposes branch flow limits then do

constraint.set_parameter('rating', 'B')

setting constraint parameters is already supported one just needs to make
the specific constraint take advantage of it

ttinoco, 2018-11-02:
see the following
https://github.com/ttinoco/PFNET/blob/dev/src/problem/constr/constr_CFUNC.c#L267
https://github.com/ttinoco/PFNET/blob/dev/src/problem/constr/constr_CFUNC.c#L14
(constraints can have their own data structure to store params)
https://github.com/ttinoco/PFNET.py/blob/dev/pfnet/cconstr.pyx#L417

ttinoco, 2018-11-02:
we need to keep in mind that are several thermal limit-related constraints
ac flow limits
dc flow limits
linearized ac flow limits

ttinoco, 2018-11-02:
> Aw ok. That looks good to me. I may see if I can quickly add this while I
> work on the bounds constraints for VSC and FACTS.

we can also split it, let me know and I can tackle either one (with
unittests it will take some time)

awig, 2018-11-02:
@ttinoco Do you have a specific way you build the tarball or is it as simple as zipping up the PFNET folder?

Thanks,

Adam

ttinoco, 2018-11-02:
If you are in linux you just do "make dist" in the root folder of PFNET (C
library), then move the .tar.gz file over to the pfnet.py repository under
the "lib" directory.
If you are in windows there is a ".bat" file in romcon PFNET "new
components" (dist_lib.bat), but the resulting .tar.gz does not work on
linux so we should not commit the .tar.gz file when created in windows